alexqzd/Marlin-H32 #2: Alex H32 Beta installed but display is garbled

tomlai-git, 2022-02-28:
garbled interface ... installed the default no probe bin and the new dwin set

[image]
[image]
[image]
[image]
[image]

tomlai-git, 2022-02-28:
Sorry NY BAD    DWIN inside the DWIN subfolder (newbie error)

codeput, 2022-02-28:
Please close the issue

closed 2022-02-28 by tomlai-git