alexqzd/Marlin-H32 #8: Print screen stuck on particular G-Code

AnshumanFauzdar, 2022-03-10:
I am printing over 24 hour with Beta1 firmware and working great so far as mentioned in issue #6 

Here is the G-Code attached [With image preview script it causes screen freezing but without preview it prints normally]

https://anonfiles.com/P8Scj9N3x8/CE3_Version_3_Winder_CASING_STANDARD_high_gcode

alexqzd, 2022-03-12:
Try with this version of the Cura plugin an re-slice the file
https://github.com/alexqzd/Marlin/blob/5d4286e1a80797d4c11041149063a5e1a53b7c11/Slicer%20post-processing%20scripts/Cura_JPEG_Preview.py

AnshumanFauzdar, 2022-03-16:
Thanks @alexqzd working great!
Just curious as a developer, what exactly is different in this script?

alexqzd, 2022-03-16:
I reduced the JPEG encoding quality to produce smaller images

closed 2022-03-16 by alexqzd

AnshumanFauzdar, 2022-03-17:
Thanks @alexqzd !