alexqzd/Marlin-H32 #9: Octoprint - some blobs/zits

whiney1, 2022-03-11:
[image]
Had some issues with lots of blobs on a benchy after upgrading to this firmware. Tweaked a few settings thinking that it might be extruder related, enabled coasting etc. Fixed most of them, but thought I'd try removing my octoprint setup and see if that was the overall issue - seems to be.
Photo attached is the same print, same settings, one right after the other - one by octoprint, one by SD card.

My setup:
Aquila x2 H32 flashed with Alex firmware/GUI files
 - no changes to stock extruder setup besides a slightly bigger fan on hotend
Octo4A running on Redmi Note 4, connected with USB cable via usb hub.

More an issue report than a complaint. Thanks very much for your work so far!

cuesir, 2022-03-16:
The issue is because of buffer size and baud rate. I've dealt with this for 6 months and finally found the fix to it. If alex can release a firmware with 250000 Baud with 32/64 Block_Buffer_Size / BufSize / TX_Buffer_Size

I've compiled a Marlin 2.0.9.3 for my ender with those setting changes, dramatically reduced my Zits and Blobs printing from octoprint. Used to be everywhere, now i get 1-2 zits. The serial data transfer could not keep up with the amount of commands octoprint is sending out causing delays on your print head hence the blob. If you print from your SD card, it'll be fine.

I Disabled SD support since I mainly use octoprint for enough memory to increase all these buff size. Since H32 uses some expensive paid software to do the firmware, I'm hoping I could get some support firmware for this @alexqzd

alexqzd, 2022-03-16:
I did change those settings in the main repo but some people started having serial communication issues and some were complaining about the gcode queue getting emptied early and the nozzle being cooled off before the actual print was finished. So I changed it back but kept it on my personal config, it's hard to keep everyone happy lol. Can you check if the issue still happens while using the MeatPack gcode compression plugin without having to increase the baudrate?

cuesir, 2022-03-16:
Oh no no definitely not blaming you or anything. I appreciate the work you've done. I've been using Beta 1 and so far haven't had any issues. Are you able to put in a compiled .bin with larger 250000 Baud / 32/64 Block_Buffer_Size / 32/64 BufSize / 32/64 TX_Buffer_Size?

whiney1, 2022-03-21:
[image]

I've updated to Beta 2 and am now having no issues.  I did have to redo my octoprint installation so can't definitively say this is what fixed it, but it's working well now so I'm closing the issue.

Picture is the same Gcode run via octoprint.

whiney1, 2022-03-21:
I should say, there are a few bumps but pretty sure they are just layer change related, you can see them in the lower part of the 'bow'

closed 2022-03-21 by whiney1