alexqzd/Marlin-H32 #11: Pause, power off, power on, then resume does not home z-axis

SpacebotMarcel, 2022-03-25:
I pause with the button in the menu and shut the printer down.
When I restart it, the printer detected a power outage and asks if I want to continue printing.
After heating, the x- and y-axis are homed, but z-axis is not and therefore it starts printing in the air.

Mik-S-UMS, 2022-03-26:
If it homed Z then it will crash into what is already been printed.

SpacebotMarcel, 2022-03-26:
Ok, but then there is no way I can resume printing this way. A solution would be to modify pause, so it stops on the same height or to have the option to change the position of the z-axis manually to the correct height.

Mik-S-UMS, 2022-03-26:
Pause does stop the printing where it is. If power is lost it will lose its position but assuming the Z has not changed then it can continue when power is restored, it just needs to home X+Y and Z will be set at the last position saved. 

You are not meant to turn off the printer midway though the print. This is a power failure recovery that lets you recover from a short term power outage. If the bed cools down too much the print will become detached from the bed making it impossible to continue.

SpacebotMarcel, 2022-03-26:
For me pause moves the head to parking position which is higher than the printed layer.
I know it is probably not made to be used that way, but why not? For me the prints do not get detached. So there is no real reason it should not work, except that I cannot get the head back to the correct layer height.

Mik-S-UMS, 2022-03-27:
It will move up to prevent a blob appearing on the print or melting it, and is usually used to quickly fix something that is about to go wrong or to change filament. it will return to the correct height when resumed. If you turn it off while paused is does not know it had been paused so continues to print at the same height.

If you want to stop mid print it would be better to just turn off but stopping and resuming prints is not a good idea.

Also resuming the print will start from the beginning of the last layer. If you stop it half way though it will go over what it has already printed when resumed leaving a thicker layer.