alexqzd/Marlin-H32 #15: newly flashed machine prints high, rapids to corner

jahnj0584, 2022-04-03:
Flashed it on a nice sunday afternoon. Flashed the H32 version bin on the printer, then the display set onto the screen. Machine moves properly and I created my manual mesh. Yes, I saved the settings and also tried this from stock settings.

When I send any print, it now ignores my initial priming line from Cura, and just sends it all the way back to Y220 and goes to the far right corner. The readout on the screen says it is +4.75 while pushing out filament as fast as I've ever seen.

What the heck is going on? The code worked 100% fine on stock firmware!

alexqzd, 2022-04-13:
This firmware supports both relative and absolute distances. Check that the mode set in the start and end gcode matches the mode set in the slicer.

jahnj0584, 2022-05-02:
I guess that was it, sorry.