alexqzd/Marlin-H32 #18: BLTouch not triggering Z axis stop

JayJay6850, 2022-04-09:
Original Printer: Aquila S2
Chip: HC32FA60
Firmware: Alex's - BLTouch-4x4-HS-H32.bin
Probe: Antclab's BLtouch SMART V3.1
Hot end (not sure it matters): Haldis 3D Red Lizard Q1 V6
After installing the probe, and disconnecting the z end stop from board, I powered on the printer. The BLtouch has lights. I wanted to test the Zstop function away from the bed (several inches), so I auto-home (tried several different menus to home, including in the z offset, and homing menus). X and Y homes successful. BLtouch pops up and down a couple times. Normally red lit when not in use, when probe is extended in the red light is off, blue still on, and the gantry begins lowering. I placed my fingernail (attempted with other items and the bed itself once) under the BLtouch and allow it to lower, pushing the probe up. The BLtouch doesn't "pop" up until the last 1mm if it does at all. gantry continues down. BL is flashing red light sometimes and sometimes just solid red. Probe never extends again. Eventually gantry stops and goes up and down a couple times or continues towards the bed. Then LCD displays that I need to restart due to home unsuccessful or I turn power off to ensure the nozzle is slammed into the bed. Any ideas what I am missing/ doing wrong?

Archer4499, 2022-06-12:
I had a very similar problem with the CR Touch on the original Aquila and, following some suggestions, tried reflashing Alex's firmware a few times and trying different versions of this firmware didn't help.
I then tried flashing Voxelab's BL Touch firmware to check if that worked, which it did. (I left the display firmware on Alex's version so it was a little tricky to see what the menu options were, but I found the homing option which was enough to test it)

I thought I might try one more time with this firmware.. and it worked. I have the BLTouch-5x5-H32 version installed and it has been working perfectly for a while now. I'm sure the other versions would also work, but I've got it working now and I don't want to ruin that.

I don't know if it'll work for others, but it's maybe worth trying to install Voxelab's firmware and then switching back to Alex's version. Maybe it overwrites/cleans a spot in memory that wouldn't otherwise be when switching between Alex's versions?

(I also have a video of it not working if that might be handy for debugging?)

JayJay6850, 2022-07-14:
That would be awesome! I wonder if flashing Voxelab's BL touch creates a proper base for Alex to launch from? I will give this a shot, thank you

ckier, 2022-10-23:
I had exact same experience as Archer4499.

hanscts, 2022-12-20:
Thanks @Archer4499 for the solution. The exact same thing happen to me and flashing the Voxelab firmware first worked. Nearly went and refunded my CR Touch.

Po5tup, 2023-02-28:
> I had a very similar problem with the CR Touch on the original Aquila and, following some suggestions, tried reflashing Alex's firmware a few times and trying different versions of this firmware [...]

Archer thank you so much man I was so close to returning it or swapping motherboards.