alexqzd/Marlin-H32 #29: Aquila X2 removal of the temperature limit from 260 to 300

NurikNar, 2023-02-13:
Hello. I would like to ask you for help. If anyone else can help I would be very happy. I want to say in advance that you have made a great contribution to custom 3D printing, and I am very grateful to you for this.
I state the essence of the problem.
I have an Aquila X2 3D printer with H32 motherboard.
Yes, I used your ready firmware .bin file. But there is a limitation on the maximum heating of the heating nozzle by 260 degrees, I would like to increase this value to 300, since my Extruder allows me to heat up to this temperature. Can someone send the finished firmware with this setting ???
I wanted to make a firmware for this printer myself, but I can’t assemble the firmware (Marlin) for this board, since there is no such option in the list of motherboards (chips). There are on G 32 and N 32. If you compile for these boards or some suitable ones, then a black screen comes out after the printer is flashed.
I'm sorry, maybe I'm asking stupid questions, but I started to study this topic relatively recently.
Oh yes and more. I flashed your Marlin configuration with my Aquila s 2 settings and everything worked for me.
Thank you for that. I hope someone can help me with my problem with Aquila X2. Thank you all in advance.
nuriknar2016@gmail.com

Mik-S-UMS, 2023-02-13:
If you are trying to do this on windows it will give an error due to a limitation  (Command prompt cannot handle the length of some strings resulting in error 87)

There are a couple of guides on reddit on how to do it in a VM using Linux. One of these may help.

https://www.reddit.com/r/VoxelabAquila/comments/wmeoty/how_to_compile_custom_alex_firmware_h32_board/
https://www.reddit.com/r/VoxelabAquila/comments/u14l60/compile_h32_alex_firmware_in_windows_on_eclipse/

NurikNar, 2023-02-13:
> If you are trying to do this on windows it will give an error due to a limitation (Command prompt cannot handle the length of some strings resulting in error 87)
>
> There are a couple of guides on [...]

[image]

and then what to choose from this list for compiling the firmware ???  for H32 chip??

Mik-S-UMS, 2023-02-13:
Sorry I don't know, I have not done it myself. 

From those guides it seemed to be already set up with the correct chip in the H32 repository.

I don't know if this is the one but there is a [hc32f46x.svd](https://github.com/alexqzd/Marlin-H32/blob/main/firmware/hc32f46x.svd) file in Marlin-H32/firmware/

NurikNar, 2023-02-13:
thanks for answering my questions.  it's important for me.  I also saw this file with all the folders, but I did not understand how to use them and this file.  how to use it?  in some folder to throw where Platformio is installed?  Or in a folder with Marlin?